plasmapotential/HEAT #5: AppImage missing FreeCAD

kosl, 2021-12-08:
Can the AppImage include FreeCAD and fix the following:
~~~bash
/tmp/.mount_HEAT_AqoH3hc/usr/src/dashGUI.py:132: UserWarning:
The dash_html_components package is deprecated. Please replace
`import dash_html_components as html` with `from dash import html`
  import dash_html_components as html
/tmp/.mount_HEAT_AqoH3hc/usr/src/dashGUI.py:133: UserWarning:
The dash_core_components package is deprecated. Please replace
`import dash_core_components as dcc` with `from dash import dcc`
  import dash_core_components as dcc
/tmp/.mount_HEAT_AqoH3hc/usr/src/dashGUI.py:139: UserWarning:
The dash_table package is deprecated. Please replace
`import dash_table` with `from dash import dash_table`

Also, if you're using any of the table format helpers (e.g. Group), replace
`from dash_table.Format import Group` with
`from dash.dash_table.Format import Group`
  import dash_table
QIconvCodec::convertFromUnicode: using Latin-1 for conversion, iconv_open failed
FreeCAD 0.18.4, Libs: 0.18.4R
No modules found in /tmp/.mount_HEAT_AqoH3hc/usr/lib/freecad-python3/Mod
During initialization the error No module named 'freecad' occurred
Directory  /home/ITER/kosl/HEAT/data/tmpDir/  Created
Select a machine before loading MHD parameters
Initializing HTML PFC data storage object
Initializing HTML PFC data storage object
Select a machine before uploading input file
Initializing Data Store
Machine not in machine list
Updating Session Store
Initializing Data Store
Machine not in machine list
Updating Session Store
...
~~~

plasmapotential, 2021-12-08:
Hey @kosl.  Thanks for raising this issue.

The appImage should include all of the necessary freecad bins / libs.  I have tested it on Ubuntu and Centos.  Which OS and version are you running on?  Does the appImage crash or continue to run?  Can you load a CAD .STEP file in the CAD box?  If this issue prevents you from running HEAT I can try to compile a new appImage for you.

Also, FYI, the latest appImage is slightly different from the tutorial videos.  I will be creating new tutorial videos later this month.

Thanks,
Tom

kosl, 2021-12-08:
I am able to follow the tutorial on CentOS 8.2. Everything seems to work except getting no results from "openFOAM thermal analysis" selected resulting empty openFoam/heatFoam/allFoams directory.  
See [output.log](https://github.com/plasmapotential/HEAT/files/7678183/output.log) attached.  Maybe something with "Could not make plotly directory" or some GUI crash using Chromium browser
~~~
Traceback (most recent call last):
  File "/tmp/.mount_HEAT_AqoH3hc/usr/src/dashGUI.py", line 2123, in loadOF
    gui.loadOF(OFstartTime,OFstopTime,
  File "/tmp/.mount_HEAT_AqoH3hc/usr/src/GUIclass.py", line 2085, in loadOF
    self.OF.meshMinLevel = int(OFminMeshLevel)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Traceback (most recent call last):
  File "/tmp/.mount_HEAT_AqoH3hc/usr/src/dashGUI.py", line 2123, in loadOF
    gui.loadOF(OFstartTime,OFstopTime,
  File "/tmp/.mount_HEAT_AqoH3hc/usr/src/GUIclass.py", line 2085, in loadOF
    self.OF.meshMinLevel = int(OFminMeshLevel)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
~~~

kosl, 2021-12-08:
The GUI errors went away by loading defaults first that set OpenFOAM min/max mesh level. I am still missing temperatures. Any suggestion?

plasmapotential, 2021-12-08:
The allFoams directory does not contain anything for now (future version will).  The .foam file lives in the openFOAM/heatFoam/<part#>/ dir.  Is your simulation generating a .foam file in the following location?
~/HEAT/data/<machine_timestep>/openFoam/heatFoam/<part#>/

where <machine_timestep> is your machine tag and timestep (ie nstx_204118), and <part#> is the part number from the CAD.

If there is no .foam file in this location let me know and I will troubleshoot with you.  If there is a .foam file in there it should contain temperature when you load it into paraview.  If the .foam file exists and there is no temperature in paraview, check to make sure the following is all true: 

In the "openFOAM Settings" box:  
1. The timestep of the GEQDSK must be between the "Start Time" and "Stop Time".  
2. The "deltaT" and "Write deltaT" must be fine enough to resolve the GEQDSK timestep.  HEAT interpolates the STL heat flux onto a volume mesh at each deltaT interval, and then solves the heat diffusion equation to generate temperatures.  So if your GEQDSK falls on a timestep that is not a multiple of deltaT, then it will skip over it and not assign heat flux.  So the temperature would be 0.  For example, to resolve a GEQDSK timestep of t=9ms, you would need deltaT=9ms or deltaT=3ms or deltaT=1ms.  This is a bug that is on the HEAT punch-list for the next revision, but you should be able to work around by changing deltaT.

If none of this works, or if there are still some problems, let me know.  I would also be happy to have a Zoom call and guide you through your first thermal analysis run.  My email is tlooby@vols.utk.edu if you want to set that up.  I am happy to help via chat, email, or zoom.
-Tom

kosl, 2021-12-13:
https://user-images.githubusercontent.com/4527143/145786620-fee46900-dd35-4bef-9f58-0b23729ca278.mp4
Video uploaded shows steps taken without getting Foam ouput. Used default eqdsk@4ms and for that write timestep was adjusted to 1ms. Is this Appimage or settings problem?

plasmapotential, 2021-12-14:
Hey @kosl.  Good idea sending me a video.  That was very convenient.  Your results make sense.  It looks like the problem is in the setup for the run, not the appImage.  You have everything working correctly, so good job getting HEAT installed.

The reason you get the line: "Not including SOLID843 in thermal analysis" is because you have selected SOLID843 to be a "Gyro Orbit Power Source" (you do this at ~2:20 in your video) in the gyro-orbit settings.  HEAT uses special source planes that a user can put in the CAD to calculate gyro-orbit heat loads on _other PFCs_.  So the gyro source planes are intentionally omitted from the temperature calculation.  

I suggest that you do not "Load Gyro Settings" if you are just calculating the optical heat flux.  You can just skip that box (Gyro Orbit Settings) entirely.  The videos I am going to add to the tutorials page later this month / early next month will guide you through running gyro-orbit heat load calculations in HEAT.  My apologies if this was confusing.  I will try to make the GUI more intuitive to prevent this in the future. 

On an unrelated side note: you may notice that when you run HEAT you get heat fluxes on the sides and bottoms of the PFCs.  This is because HEAT traces magnetic fields in any direction away from a PFC surface.  To constrain the heat flux to only be on the top, you would need a backing plate, another PFC, a plane, or some other CAD object.  I did not include this in the test case provided on github, but I can provide that if you think it would be useful.

Try running without any gyro-orbit settings and let me know what you get.
-Tom

plasmapotential, 2022-01-20:
@kosl if you have any further questions, feel free to email me at loobytp@ornl.gov and I will answer.  I am also open to having a meeting if that would be beneficial.  Lastly, your work with SMITER was an inspiration for our team to develop HEAT.  If you are interested in a side by side code comparison let me know.  A recently published HEAT paper is available here: https://doi.org/10.1080/15361055.2021.1951532

-tom

closed 2022-01-20 by plasmapotential