plasmapotential/HEAT #10: Setting user defined heat flux mode when in running batch file

maaarten, 2022-04-20:
For some reason I can't get the user-define heat flux mode to work when running HEAT using a batch file. When I run a specific step and PFCinput file in the GUI and select "User Defined" heat flux width source, I get a different result from when I try to run the same inputs using a batch file. When I save settings into input file after setting the heat flux width source to user defined in the GUI, the hfMode comes up blank in the input file, so I thought that's how it was meant to be for the batch file. 

I've tried both leaving the hfMode blank and setting it to "user" in the input file but both times it seems to revert to default settings and using Eich15, hence giving me different results from running it from the GUI. 

Am I doing something wrong (quite likely) or is using user defined hfMode not supported when running in batch-mode?

plasmapotential, 2022-04-20:
this may be a bug in HEAT, and the latest development version addresses it.  Are you running in the appImage or in the docker container?  

For reference, your HF settings in the input file should look like this:
```
#==============================================================
#                Optical HF Variables
#==============================================================
hfMode, eich
lqCN, 2.5 
lqCF, 5.0
lqPN, 1.0
lqPF, 1.0
lqCNmode, user 
lqCFmode, None
lqPNmode, None
lqPFmode, None
S, 0.15 
SMode, user 
fracCN, 0.6
fracCF, 0.4
fracPN, 0.1
fracPF, 0.0
fracUI,0.1
fracUO,0.4
fracLI,0.1
fracLO,0.4
Psol, 10.0
qBG, 0.0
fG, 0.37
```

this will set lambda_q to be 2.5 mm and S to be 0.15 mm.  Change the values on those lines to reflect your desired values.  The 'user' string after lqCNmode sets the mode for the near SOL lambda_q.   I will leave this thread open until a) I fix the "save to file" feature creating a blank hfMode, and b) check that this works in the next release appImage.

maaarten, 2022-04-21:
I am running in the appImage. I've edited the input file now and re-running the batch file to see if it works. I think I had miss-interpreted hfMode with lqCNmode! hfMode coming up as empty is still an issue when saving but hopefully now I can run the batch file properly. Thanks for the help, will let you know if it works.

maaarten, 2022-04-22:
Can confirm it works. Thanks for the help!

plasmapotential, 2022-04-22:
glad it worked.  I will leave the issue open until I fix the hfMode save to file bug.  Thanks for the feedback and opening an issue.

maaarten, 2022-05-11:
Hello, 

I've been looking into this a bit more after successfully running a few batch runs and realising that I am getting different results between GUI and TUI runs with what I believed were the same inputs for lq. Again, most likely I am doing something wrong. 

If I input the following numbers in the GUI: 
[image]

And then save as input file then they are save as below: 
#=============================================================
#                Optical HF Variables
#=============================================================
hfMode, 
lqCN, 3
lqCF, 5
lqPN, 1
lqPF, 1
lqCNmode, 
lqCFmode, 
lqPNmode, 
lqPFmode, 
S, 1
SMode, 
fracCN, 0.6
fracCF, 0.4
fracPN, 0.1
fracPF, 0
fracUI, 0.1
fracUO, 0.4
fracLI, 0.1
fracLO, 0.4
Psol, 10
qBG, 0
fG, 

Where lqCN is 3 and the "User Defined Heat Flux Width [mm]" I set to 2.5 is not there, and the lqCNmode is empty (as is hfMode), but an S of 1 is included. When I run the simulations the GUI results look like they had a wider lq (eg less heat flux) than the TUI results where I set lqCN as 2.5. 

When I save the GUI input as a a file then this is printed in the terminal: 
{'MachFlag': 'st40', 'shot': '918', 'tmin': 0, 'tmax': 5000, 'nTrace': '20', 'gridRes': 'standard', 'lqEich': '2.5', 'S': 1, 'lqCN': 3, 'lqCF': 5, 'lqPN': 1, 'lqPF': 1, 'fracCN': 0.6, 'fracCF': 0.4, 'fracPN': 0.1, 'fracPF': 0, 'Psol': 10, 'fracUI': '0.1', 'fracUO': '0.4', 'fracLI': '0.1', 'fracLO': '0.4', 'qBG': 0, 'OFtMin': 0, 'OFtMax': 20, 'meshMinLev': 1, 'meshMaxLev': 2, 'STLscale': 1, 'deltaT': 0.001, 'writeDeltaT': 0.001, 'FreeCADPath': '/tmp/.mount_HEAT_AIHSA83/usr/lib/freecad-python3/lib', 'PVPath': '/tmp/.mount_HEAT_AIHSA83/opt/paraview/lib/python3/dist-packages/paraview', 'dataPath': '/home/erik_maartensson/HEAT/data', 'OFbashrc': '/tmp/.mount_HEAT_AIHSA83/opt/openfoam/etc/bashrc', 'N_gyroSteps': 5, 'gyroDeg': 10, 'gyroT_eV': 100, 'N_vSlice': 1, 'N_vPhase': 1, 'N_gyroPhase': 1, 'ionMassAMU': 2.014, 'vMode': 'single', 'ionFrac': 0.5}
OFtMin
OFtMax
deltaT
writeDeltaT
STLscale
meshMinLevel
meshMaxLevel
material
Wrote HEAT input file

Where I haven't actually seen the term "lqEich" before, and lqCN is still set to 3. So how does this compare to running the same values in TUI mode?

plasmapotential, 2022-05-11:
Thanks for the updated comments.  I believe this is related to a bug in heat that was addressed (after our previous discussion) in this commit:  https://github.com/plasmapotential/HEAT/commit/0cc695087e9bcb89b8ca7cecc9d2f1d196b74216

I am waiting to close this issue until the release is out.

Basically, there were some legacy parameters in the GUI that were overriding the user's settings.  This new version of the HEAT gui has been cleaned up, and there are some new features (ie loading HF from a file).  It will be released within the next couple of weeks.  If you need access to this new code now, you can run HEAT via the docker container and pull the HEAT source code from github, as described in the tutorials (and Chris has also done this).  If you opt for this route, let me know and I will merge my development branch with master.

I suspect the TUI is correctly calculating lambda_q, and I would trust TUI over GUI for the v2.0.0 release.  You could test by setting the lq to something large and running through the TUI to make sure your change to the input file is propagating to the results.

If you want a meeting to discuss, feel free to reach out and we can set something up.  I am cleared to assist all third parties with HEAT related topics, even if they are private companies external to CFS.

plasmapotential, 2022-05-11:
I just patched a few more bugs and merged with master.  running in docker should now give you access.  Note that the latest HEAT push is using the latest version of dash / plotly, and the docker container uses and older version.  you may need to updgrade dash / plotly in the docker container before running HEAT

maaarten, 2022-05-16:
Hi Tom, 

Thank you for making those changes. I haven't run the docker yet, will see how much time I have in the next few weeks, will either sort out the docker container or just run it when the new release is out. Cheers

plasmapotential, 2022-11-04:
patched and tested in HEAT v3.0

closed 2022-11-04 by plasmapotential