plasmapotential/HEAT #12: changing "Data Directory" is not working

msubraman, 2022-07-14:
Hi,

I am running HEAT in WSL, data directory is not changing and data are stored in same directory (/home/'username'/HEAT/data/) even after I mention new data directory in GUI. I want to store the data into other drives like "/mnt/e/...".
Can you please help to solve this issue.

Thanks

plasmapotential, 2022-07-14:
Hello,

Currently, HEAT uses the "~" directory for the data path, as shown here:
https://github.com/plasmapotential/HEAT/blob/e287ae7a8726d517e405f4dbf8486c3e7a406cb4/source/launchHEAT.py#L30-L38

For now, you could try to:

- create a symbolic link between /mnt/e/... and $HOME
- export your desired path into $HOME before running HEAT: (ie export $HOME=/mnt/e/...)

I should mention that it has been requested to also allow a user to define another variable, $HEATHOME, before running HEAT  for this purpose, but this will not be available until the next release of HEAT.

Lastly, HEAT running on WSL has worked a few times, but is relatively untested.  The best method for running on Windows is inside a docker container (which also runs on WSL, but includes the entire HEAT environment inside the ubuntu container).  If you are already running in docker, then that is good.

Let me know if you have any other questions, and keep an eye out for the next release with the $HEATHOME variable.

plasmapotential, 2022-07-14:
I will also add a patch to the next release that allows user to define data directory from GUI.  Thanks for reporting this bug

msubraman, 2022-07-14:
Thanks Tom for the reply.
I will try the export $HOME="" command. I think it should work.

I am running HEAT wsl, not tried docker but I will try docker container. As of now I haven't faced any no issues running HEAT on wsl. 

And I am waiting for the next release.

msubraman, 2022-07-15:
Hi Tom,

It works with the command,
"export HOME=/mnt/....."

Thank you and have a good weekend.

closed 2022-07-15 by msubraman