plasmapotential/HEAT #14: New version of heat not working properly

msubraman, 2022-11-09:
Hi Tom,

Manoj from UKAEA.
I followed the steps for the installation of HEAT with docker (v3). Everything went well but the GUI is not working well. 
GUI looks different (picture attached) as well as giving errors while loading the files (input, eqdsk,..) 
I have also included docker-compose.yml and runDockerCompose

[image]
[image]
[image]
[image]
[image]

plasmapotential, 2022-11-09:
Hey Manoj,

Thanks for submitting this issue.

I think you are missing the radiated power settings in your input file.  The Radiated power section of the input file should look like this

```
   #==============================================================
   #                Radiated Power HF Variables
   #==============================================================
   radFile, /home/tom/HEATruns/testCase_v3.0.0/nstx/RZpower.csv
   Ntor, 5
   Nref, 1
   phiMin, 20
   phiMax, 35

```

Could you upload the NSTXU_input.csv file that you are using?  Is this the NSTXU_input.csv file that came from the HEAT releases tab (testCase_v3.0.zip) ?  https://github.com/plasmapotential/HEAT/releases/tag/v3.0

plasmapotential, 2022-11-09:
One more comment.  The GUI theme can now be changed in HEAT in the top right corner using the "Change Theme" button

msubraman, 2022-11-11:
The problem still exist after trying both ways as mentioned above.
But I found a solution using 'export HOME=/mnt/...' , that's working and I can open the results.

My results gyro results are not showing any heat flux mapping, so I think it's lack of knowledge for me to use the gyro parameters. To understand better the testcase for gyro/photon and new features, I think we can meet and you can guide me through. I will send you an email.

plasmapotential, 2022-11-12:
ok I think in a meeting we will clear this up.  One other thing you could try is to install the latest version of docker.

I have been meaning to update all the youtube tutorials and to include a youtube video for both gyro orbit and radiated power, but I cannot seem to find the time to do this.  Hopefully over the holidays things will slow down and I will be able to publish these tutorials.  But yes, let's meet and I will describe everything.

I will wait to close this thread until we find a solution in our meeting.  I will post the solution here for future users.

plasmapotential, 2022-11-20:
solved by using runDockerCompose and uncommenting the "environment" section of the recipe YAML file, docker-compose.yml

closed 2022-11-20 by plasmapotential