plasmapotential/HEAT #15: HEAT v3 install - vtk ModuleNotFound

VicFusion, 2022-11-11:
Good day all,

First of all, thank you for developing this tool and keeping it up and running as open code.
I think it is fair to say I am new to this (i.e. Linux in general and using development codes specifically), but I wanted to get familiar with HEAT after hearing about it.

When trying to install HEAD v3 following the instructions the following error pops when initiating the docker container:
[image]

After this I can't connect to the GUI via web browser.
Am I missing something obvious (missing a lib in my desktop) or is this library missing from the source code (I got it from git yesterday)?

Kind Regards,
Victor

plasmapotential, 2022-11-12:
Hey Victor,

This was my mistake!  I did not push the latest version of the docker container to dockerhub after I updated the VTK module and HEAT output filestream on Thursday.  I have now uploaded the new docker container to dockerhub (v3.0.1).  You should be able to pull the latest HEAT source (git pull) and the latest docker container (docker pull) and it should work.  If not let me know.

Thanks for being here.  And thanks for submitting the bug report.  If you run into any other issues feel free to let me know.  Sorry for the confusion!
-tom

VicFusion, 2022-11-13:
Good morning Tom,

Thank you for your quick reply.
I am pulling and will try shortly.

Thank you,
Victor

plasmapotential, 2022-11-14:
No problem, Victor.  If anything else comes up, feel free to open another issue on github.  Also, the tutorials/youtube should be updated sometime this month to reflect the latest version of HEAT.

VicFusion, 2022-11-18:
Good morning Tom,

Sorry for taking this long to try again, work got in the way.
Docker is still giving me the same error when trying to run it. I have
deleted all docker containers and images and the HEAT documents and started
from scratch but pulled v3.0.1 and this is still the case.

Is there anything I can do to help you troubleshoot it?

Kind regards,
Victor

plasmapotential, 2022-11-20:
Hey Victor,

This was because i forgot to stage the docker-compose.yml file when i pushed the changes.  Pull the latest HEAT sources code (git pull) and then try again.  Sorry about that!

To check, you can confirm that your docker-compose.yml file has the line:
```
image: "plasmapotential/heat:v3.0.1"
```

-tom

VicFusion, 2022-12-06:
Good evening Tom,

Apologies for the late response, I have change the line in the docker-compose.yml to "plasmapotential/heat:v3.0.1" and it runs fine at the moment.
I am looking forwards to getting started with the tutorials.

Kind Regards,
Victor

closed 2023-02-28 by plasmapotential