plasmapotential/HEAT #18: Query about PFCs with both mapDirections in batch mode

chrismarsden7, 2023-07-14:
Hi Tom,

I had a quick query (apologies if you have already explained this before). When running on a PFC that sees both mapDirections (like the PFC at the tangent point when limited for example), normally, you would just list that PFC twice right? (Once in the one divertor region, and again in the second divertor region).

When running in batch mode, I experience issues with 'PFC has no attribute mapDirection' which only occurs for those PFCs that actually do sit in 2 different regions (like the tangent point). I suspect this is because I was listing the PFC twice (once in each region). Is this the correct procedure, or is there a specific divertor code for those PFCs that see multiple regions?

Cheers

plasmapotential, 2023-07-14:
Hey Chris.  Hope all is well.

Which version of HEAT are you using?

A couple years back after you and I discussed the problems with mapDirection, HEAT was updated so that there no longer is a mapDirection in the PFC file.  HEAT autonomously figures out that for every mesh face there is only one way power could flow into it given the magnetic field constraints (helicity, Bt direction, etc.).  So now for a limiter tangent point face, you only need a single line in the PFC file, and there is no need for a mapDirection column.  

As for the divertor codes, there is not a current method to split a single CAD object into two divertor codes.  If you are doing a inner limited pulse, then you can set the divertor code to UO and just set the fracUO to 0.5, because for simple inner limited pulses the power is likely flowing in both directions at a ~ equal ratio.  For more complex situations (like a diverted / limited pulse), we may need to be creative, or add a new feature into the code.

Here is an example of a PFC file in the current version of HEAT:
```
timesteps, PFCname, resolution, DivCode, intersectName, excludeName   
0:5000, SOLID002, 5.0, LO, all, none
```

Did that answer your question?  If not, or if you need an additional feature, could you describe the situation a bit (vague is fine).

-tom

chrismarsden7, 2023-07-16:
Hi Tom,

I am using the latest version of HEAT.  I do remember how we set up HEAT such that mapDirection is no longer specified and is now automated, hence why I was a bit surprised to see it pop up in the error message.

Yes, it was for exactly this, the central tile of for a limited discharge - what you've said in your reply makes total sense. I wasn't sure if you had said something previously about treating these kinds of tiles in a specific way - I was probably just misremembering something.

Cheers for answering my question - please consider this issue closed : )

Cheers,
Chris

closed 2024-03-21 by tloobyCFS