plasmapotential/HEAT #24: plasma3Dmask from GUI

tloobyCFS, 2024-01-11:
when running HEAT in GUI mode, the new plasma3D branch does not handle the plasma3Dmask properly and HEAT fails because the self.MHD.plasma3Dmask var is not set.  fix by initializing the variable in the GUI

wingena, 2024-01-12:
The plasma3D part of HEAT is only working correctly in terminal mode for now. Changing the GUI to enable proper handling of the plasma3D part is on the to-do list, but of lower priority than other outstanding tasks.

tloobyCFS, 2024-01-12:
@wingena the problem isn't that the user cant run the 3d plasma calculation from the GUI (im still debating if we should do that long term anyways).  The problem is that when a user wants to run an axisymmetric plasma from the GUI the variable self.MHD.plasma3Dmask is not set to False, so they cannot run a 2D case.  I have a patch for this that i will push into v4.1

tloobyCFS, 2024-03-21:
should be fixed in PR#36

closed 2024-03-21 by tloobyCFS