plasmapotential/HEAT #29: new 'facingOut' function [use] indexes not working

tloobyCFS, 2024-02-08:
I ran a series of SPARC simulations and found that the shadows were being incorrectly calculated.  More specifically, I found that the first trace step was failing to identify any shadows.  Digging deeper I noticed where @wingena added the "facingOut" variable in [findOpticalShadowOpen3D()](https://github.com/plasmapotential/HEAT/blob/f6b7d74b10e56f0bb458f2bc4cec990e80a039df/source/pfcClass.py#L399).  The [use] index is being used to reference the result from the facingOut function, and [use] indexing was added to the first trace step for self.powerDir, self.centers, etc (it was not used before).  This is all fine, but the assignment of `intersect_mask` to `self.shadowed_mask` was incorrect and failing.

This is what was coded [here](https://github.com/plasmapotential/HEAT/blob/f6b7d74b10e56f0bb458f2bc4cec990e80a039df/source/pfcClass.py#L466):
`self.shadowed_mask[use][fwdUse] = intersect_mask`
This is the corrected code:
`self.shadowed_mask[use[fwdUse]] = intersect_mask`

The reverse trace code needed correcting as well:
`self.shadowed_mask[use[revUse]] = intersect_mask`

I have fixed in this commit, which i will submit a PR for in a week or two:  https://github.com/plasmapotential/HEAT/commit/904d6fd17bb21a821f86f3faeb3ec445426b8e6b

I think there is a broader issue here, that we need an intersection check integration test.  I will work to develop an integration test that can run in CI and checks the shadowMask is calculated correctly.  Hopefully this will serve to prevent these issues (and time spent looking for bugs) in the future.

@wingena you need to correct your plasma3D branch to reflect these changes.  Thankyou for commenting the original lines but leaving them in the code.  I would not have found the bug unless the commented old lines were still there.

Leaving this issue open until the integration test has been completed.

wingena, 2024-02-21:
I updated this in the plasma3D branch using your patch. However, there are two more lines that needed the same fix:
Line 453 was:
MHD.writeMAFOTpointfile(self.centers[use][fwdUse],self.gridfileStruct)
Now reads:
MHD.writeMAFOTpointfile(self.centers[use[fwdUse]],self.gridfileStruct)

Same for the reverse trace a few lines below.

plasmapotential, 2024-03-05:
resolved with PR #31

closed 2024-03-05 by plasmapotential