plasmapotential/HEAT #30: Unable to run 2D plasmas

chrismarsden7, 2024-02-23:
Hi Tom

Just moved over to v4, and am encountering issues running HEAT on cases I have previously ran without issue. This issue seems to be related to the new 3D plasmas. I kept getting this error (below), but could not find an entry in the GUI to influence this. I then took the NSTX input file and noticed it had an entry for this mask. I modified said file, hoping it would initialise plasma3Dmask, however, that still didn't work and kept giving this error.

```
raceback (most recent call last):
  File "/root/source/HEAT/source/dashGUI.py", line 2840, in runHEAT
    gui.runHEAT(runList)
  File "/root/source/HEAT/source/engineClass.py", line 1760, in runHEAT
    if self.MHD.plasma3Dmask == 1:
AttributeError: 'MHD' object has no attribute 'plasma3Dmask'
```

Cheers,
Chris

tloobyCFS, 2024-02-23:
hey Chris,  
this issue has been flagged here for terminal mode:  https://github.com/plasmapotential/HEAT/issues/26

and here for GUI mode:  https://github.com/plasmapotential/HEAT/issues/24

there is a patch in my dev branch, but it probably wont be into main for a few weeks. to get around this for now you can just 0 out the variable just before the error line in engineClass.py:
```
self.MHD.plasma3Dmask = 0:
if self.MHD.plasma3Dmask == 1:
```

This error is a result of the new 3D fields module (reading 3D plasma equilibria from M3DC1 is now in HEAT).  There are a few bugs that we didn't catch before the release of v4.0.  To prevent bugs slipping through the cracks in the future we have created a series of integration tests in HEAT that will check the functionality of various modules when the container gets built and new pushes to main are merged.  

When I release v4.1 (probably a few weeks to a month) there will be several patches such as this one, additional integration tests, some documentation on filament and 3D plasma module in the readthedocs, and a new stress module.  I will leave this issue open until the patches are in main.

Let me know if the aforementioned workaround doesn't get you up and running.  If you are running in batch mode look at issue 26 above.

chrismarsden7, 2024-02-26:
Hi Tom,

Cheers for the quick response! I've implemented this change now. In order to test it I had to uncomment the line in the docker-compose yaml file to point HEAT towards the local source code (I normally do this, but hadn't yet done so when moving to v4):

```
#      # HEAT source code from github
      - /home/chris.marsden/GitHub/HEAT/source:/root/source/HEAT
```

however, I am now getting this error when launching the docker container

```
Recreating docker_HEAT_chris_1 ... done
Attaching to docker_HEAT_chris_1
HEAT_chris_1  | python3: can't open file '/root/./source/HEAT/source/launchHEAT.py': [Errno 2] No such file or directory
```

Any ideas?

Cheers,
Chris

tloobyCFS, 2024-02-26:
Hey Chris,

In v4.0, we changed the directory structure in the container slightly.  This is so that we can have integration tests inside the container.  The new line in the bind mount for the HEAT source code should look like this:
```
      # HEAT source code from github
      - /home/tlooby/source/HEAT/github:/root/source/HEAT
```

As a side note, you should be able to test out your installation of HEAT by running the following commands in the container:
```
cd ~/source/HEAT/source
runTerminalModeTestOptical
```
This will run the NSTXU test case with optical approximation.  When you are ready to run your own case (that you bind mounted to /root/terminal), you can use this shell script from that directory:
```
runTerminalMode
```
Of course, you can alternatively still use the old method where you manually enter the command:
```
python3 launchHEAT.py --m t --f /root/terminal/batchFile.dat
```

If you can't find the runTerminalMode scripts or launchHEAT.py, I suggest you jump into the terminal and look around to see where the files got mounted.

Let me know if you have any other questions.

chrismarsden7, 2024-02-26:
Hi Tom

Sorry if I'm misunderstanding, but this doesn't seem to have done the trick for me.

My clone of the HEAT repo is located at /home/chris.marsden/GitHub/HEAT. Within which is the source folder at /home/chris.marsden/GitHub/HEAT/source . I've changed the yaml entry to edit the path to the 'source' folder:

```
      # HEAT source code from github
      - /home/chris.marsden/GitHub/HEAT/source/HEAT/github:/root/source/HEAT
```

which didn't work. I've also tried:

```
      # HEAT source code from github
      - /home/chris.marsden/GitHub/HEAT/source/github:/root/source/HEAT
```

which also didn't work :/

Cheers,
Chris

chrismarsden7, 2024-03-06:
Hi Tom,

Thank you for the clarification - all working now! : )

Cheers,
Chris

tloobyCFS, 2024-03-21:
this should be patched in PR #36.  After it gets merged, test it out and see if it fails.

closed 2024-03-21 by tloobyCFS