plasmapotential/HEAT #32: It is possible to contribute to this project ?

cdrojasm, 2024-03-04:
If it is possible to contribute, have you some guidelines or a roadmap where people can get some information about requirements and features ?

tloobyCFS, 2024-03-04:
Hey @cdrojasm thanks for inquiring!  Yes, it is possible to contribute to the code, and there are HEAT developers around the world working on different features.  The best way to get involved would be to 1) download the code and run through the tutorials to get a feel for running the code, and 2) have a 1:1 meeting with me where we define some potential areas of collaboration.

There is not a "roadmap" document per se, but there are definitely a lot of feature requests and patches that eventually need to be added to the code.  We could probably consolidate all the feature requests into a single document or webpage, and maybe that is a good idea for long term growth.  My philosophy up to this point has been to just let people add the features they think are relevant, so the development has been driven by the community's desires rather than a well-organized roadmap.  

Is there a specific use case or feature you need / are looking to add?  Or are you looking for a more general "list of feature requests" that you could tackle to contribute to the project?

Feel free to send me an email at tlooby@cfs.energy if you want to discuss or potentially set up a 1:1.

Thanks for reaching out!
-tom

cdrojasm, 2024-03-05:
Great, I will start reading the current documentation and video tutorials. After I get a bit familiar with the project then I will contact you !!! 

Thanks a lot.

closed 2024-03-05 by cdrojasm