plasmapotential/HEAT #44: Edge point interpolation issue in fluxCoorDistance function in filament module

feyrer, 2024-07-23:
fluxCoordDistance function is unable to handle center points which are located at theta = pi, due to thetasurf not going all the way around. Have attached the fix applied in the runaway module. 
[image]

feyrer, 2024-07-23:
The above fix didn't actually work. The trouble is that thetaSurf starts and ends somewhere around -1.8 rather than -pi, pi. I have added these lines of code and that seems to work: 

        #Fixing 2pi issue
        pi_ind = np.argmin(np.abs(thetaSurf - np.pi))
        negpi_ind = np.argmin(np.abs(thetaSurf + np.pi))
        thetaSurf = np.append(thetaSurf, np.array([thetaSurf[negpi_ind] + 2 * np.pi, thetaSurf[pi_ind] - 2 * np.pi]))
        dSurf = np.append(dSurf, np.array([dSurf[negpi_ind], dSurf[pi_ind]]))

plasmapotential, 2025-01-23:
@feyrer implemented the same fix you have in runaway branch in this commit: 9f3e5015c9c8f8b78b8ec813b201434d3962a27d

closed 2025-01-23 by plasmapotential